basdelfos/beta-osd-v2 #4: Uploading firmware without resolder

basdelfos, 2016-08-28:
[Issued by @llambkin] Is there a way of uploading new FW to the OSD without having to resolder, maybe I need to flash a new bootloader so the FC can trigger a reset to the atmega, then new FW can be uploaded using xloader, and or possibly the Betaflight Configurator in the future.

basdelfos, 2016-09-01:
That would be an easier way to flash the OSD fw indeed. Don't expect this to be implemented very soon into Betaflight.

llambkin, 2016-09-01:
It wouldnt have to be a Betaflight implementation, since all BF does is
pass serial comms through to the OSD, but I have no idea how much work it
is to develop a configurator that implements a software reset to upload new
firmware.

DerOrakel, 2016-09-02:
Soldered some angled pins on my micro minimosd and hardwired rx/tx to the bottom of these pins. Works fine for flashing without resolder here.

basdelfos, 2016-09-02:
I also place some angled pins on them ala Oscar Liang's idea. It fits right after the camera and flashing a new firmware is not that hard as it's just disconnecting a few connectors. I'm even thinking about buying some 12pin header connectors, then it is just only one connector to disconnect. I will make a diagram on how I wired the pins, the upper row is 2x 3pin VIDEO/5V/GND for video in and out. And soldered in a way that the OSD gets its 5V/GND from the VTX.

[image]