basdelfos/beta-osd-v2 #5: NTSC not working and an improvement suggestion

DerOrakel, 2016-08-30:
Hi,

thanks for your OSD it's exactly what I was looking for. But the NTSC Version is not working, actually I would guess it uses PAL resolution as most of the elements are off the screen. When switching TOP/BOTTOM the elements are displayed but cut off at the top as seen on the pictures attached.

[image]
[image]

Does it use FC RSSI or do i have to connect the RSSI pad (I would if I could but XSR) so an option to use FC RSSI would be great. Maybe in the way try RSSI pad if no data use FC RSSI.

About the low voltage warning, looking at the code it seems to me the warning offsets are hardcoded, in my opinion it would be nice to use the values set in betaflight as it is possible in MWOSD. That combined with AUTOCELL_COUNT as in MWOSD would really fit your flash an go OSD.

regards,
Ron

basdelfos, 2016-08-31:
Hi Ron,

Can you try the OSD fw attached. It could be that due to the PAL resolution
it forces some elements to show off screen. Let me know if it works now,
could be that some OSD elements are still off screen tho.

This fw is designed to use as less solder pads as possible, all RSSI/VBAT
information is read from the settings in Betaflight/KISS. Connecting the
RSSI pad or VBAT pad on the OSD will not work. You will have to send the
RSSI information to Betaflight on a spare channel, here is a link how to
configure this in OpenTX and Betaflight:
https://oscarliang.com/rssi-ppm-channel-taranis/

It uses the low voltage warning settings from Betaflight. In case of KISS
FC with KISS fw the voltage warning is not working yet, need to implement
this. I will look into your suggestion with AUTOCELL_COUNT from MWOSD.

Regards,

Bas.

<https://cloud.githubusercontent.com/assets/21347918/18108563/13b19064-6f0d-11e6-81eb-a154af753ea3.png>
[image: vlcsnap-2016-08-30-23h47m04s204]
<https://cloud.githubusercontent.com/assets/21347918/18108571/1802af90-6f0d-11e6-8a19-a7f9b59cfb99.png>

DerOrakel, 2016-08-31:
Hi,

i can not really spot a difference, when using TOP/BOTTOM the elements are shown using my goggles, right at the edge but visible, but with my capture card the are still half off screen. The bottom part is still completely off screen.
Also the modes indicator always shows STAB even tough BF was in ACRO mode during the capture off the screenshots.

See some pictures below.
[image]
[image]
[image]

As for the RSSI part thats excatly what i'm doing.

regards,
ron

basdelfos, 2016-09-01:
Thanks. I will look into this later today. Wanted first to be sure the NTSC is working. The specs of the MAX chip specify that NTSC has less horizontal character rows than PAL. I find it strange tho, that the first row is not fully visible, I would expect that the first row is always be visible and the that the last rows would fall off screen. Solution would be that on NTSC the first row is shift down by 1 and the bottom rows shift up by 2 rows up.

I didn't notice the modes indicator does not work any more as I don't use it atm. I probably broke it while implementing the KISS FC compatibility. I will make a new issue for that and will fix that also.

basdelfos, 2016-09-01:
@DerOrakel can you try this version. NTSC should be fixed, all rows are visible on the test 5" screen that I have.

[firmware.hex.zip](https://github.com/basdelfos/beta-osd-v2/files/450838/firmware.hex.zip)

basdelfos, 2016-09-01:
Modes indicator is also fixed in this version.

DerOrakel, 2016-09-01:
@basdelfos this version looks very good. Did not fly with it but on the bench everything is where it should be. Hopefully i can fly a few packs on Friday, but as of now i would say bug fixed.

Thank you very much.
Got another bug but i will open a new issue for it.
Also gut another idea but gonna open a new issue for that either.

regards,
ron

closed 2016-09-02 by basdelfos