basdelfos/beta-osd-v2 #7: Bug, Low Voltage Warning

DerOrakel, 2016-09-01:
Hi Basdelfos,

noticed this today, with a fully charged 4S battery plugged in the OSD shows 5S on the initialization screen and the voltage indicator constantly flashes at 16.7v. Battery settings in BF are all stock 3.0.0 RC12. 

Tested with the NTSC fixed hex and Betaflight 3.0.0 RC12 on a RCExplorer.se F3FC (the board not the mini tricopter version).

regards,
ron

basdelfos, 2016-09-02:
I have also noticed this with a full battery, seems to not do this always. I think the cell count calculation needs some fine tuning, somehow it detects a 5S battery sometimes. The voltage indicator starts flashing because the OSD thinks the voltage is to low for a 5S battery (low voltage: 5S x 3.4 = 17V).

basdelfos, 2016-09-02:
The cell calculation is correct, but i think that the false cell calculation is caused by voltage reading spikes by the FC directly after connecting the battery. To fix this, I have implemented a delay in the cell calculation of 1.5 seconds. So after 1.5 second after powerup the cells are calculated, that should be enough time to level out the voltage reading by the FC.

Could you test this firmware version: 
[firmware (cells fix).hex.zip](https://github.com/basdelfos/beta-osd-v2/files/452897/firmware.cells.fix.hex.zip)

DerOrakel, 2016-09-02:
@basdelfos that seems to do the trick. Plugged in different batteries several times worked fine every time.

DerOrakel, 2016-09-02:
Works fine with empty batteries to for me

basdelfos, 2016-09-02:
Thanks, we can consider this one solved too

closed 2016-09-02 by basdelfos