basdelfos/beta-osd-v2 #10: Dynamic position bug

DerOrakel, 2016-09-11:
Hi @basdelfos 

Using the latest commit when the mode indicator is enabled and battery voltage gets to the warning level  and starts to blink the mode indicator jumps left and right whenever battery voltage blinks out and in again.

[image]

Autocam works nice here with a HS1190 (Foxeer Arrow) ntsc.

As for the order of indicators i would prefer the old order: RSSI, VBAT, FTIMER, MODE

regards
ron

basdelfos, 2016-09-11:
Yeah, I already noticed the same issue in one of my flights during the weekend. When the voltage is below the low voltage warning threshold, it starts blinking. The blinking causes the voltage not to display and the mode is placed on the position of the voltage. 

I already fixed it earlier this evening, like 15 minutes ago :smile:. Look in the development branch.

Good to know that autocam works, the sourcecode is left-over from the original MWOSD code and I didn't test it yet.

I will fix the order, probably changed it on accident while adding the dynamic position feature.

DerOrakel, 2016-09-11:
Great compiled and flashed latest development commit and will test it monday.

basdelfos, 2016-09-11:
Actually I made a bug in mistake (should have tested it...). It is fixed now in de development branch.

DerOrakel, 2016-09-12:
Seems to work fine, not sure about the flied order tought I flashed the latest Dev commit but order still was  rssi, ftimer, vbat (mode was disabled this time). Have to take a look at it again.

basdelfos, 2016-09-12:
Strange, just tested the latest development branch and the order is RSSI, VBAT, TIMER, MODE.

[image]

DerOrakel, 2016-09-13:
Using the latest commit field order is as it was originally, so maybe my last flash went wrong or i was just making up things. I don't know it's correct now :)

closed 2016-09-13 by basdelfos